seeing-things/track #11: Takes over 10 seconds to construct VideoCapture object

bgottula, 2017-06-10:
Constructing the `VideoCapture` object for optical tracking can be really slow. I clocked it at 13+ seconds using a Logitech Quickcam IM model V-UBB39. Based on my experiments in #10, it may be that the constructor waits for the capture buffers to fill up before it returns. With my configuration the frame rate was around 15 fps and it was buffering around 161 frames, which is about 10.7 seconds of buffer time.

jgottula, 2017-08-13:
It looks like opencv, in `cv::VideoCapture::open`, first attempts to do direct MJPEG capture via `cv::createMotionJpegCapture`, but this fails, and then it falls back onto `cvCreateFileCaptureWithPreference`, which uses FFMPEG; and FFMPEG, lacking any information about the input format to expect (as opencv doesn't provide any such info), decides to buffer a ton of frames so that it can deduce the input format before it concludes the opening process.

It may be that opencv's MJPEG capture code only works if the MJPEG encoded video is in an RIFF container, which may not be the case with our particular webcam, or perhaps all webcams, for all we know. May warrant further investigation, depending on how much time we want to spend debugging this garbage.

MJPEG attempt:
```
#0  0x00007f9b797ba02d in read () at /usr/lib/libc.so.6
#1  0x00007f9b79750598 in __GI__IO_file_underflow () at /usr/lib/libc.so.6
#2  0x00007f9b7974f608 in __GI__IO_file_xsgetn () at /usr/lib/libc.so.6
#3  0x00007f9b79743eb1 in fread () at /usr/lib/libc.so.6
#4  0x00007f9b7106111e in cv::MotionJpegCapture::parseRiff(cv::MjpegInputStream&) () at /usr/lib/libopencv_videoio.so.3.2
#5  0x00007f9b7106170d in cv::MotionJpegCapture::MotionJpegCapture(cv::String const&) () at /usr/lib/libopencv_videoio.so.3.2
#6  0x00007f9b71061816 in cv::createMotionJpegCapture(cv::String const&) () at /usr/lib/libopencv_videoio.so.3.2
#7  0x00007f9b7104dc41 in cv::VideoCapture::open(cv::String const&, int) () at /usr/lib/libopencv_videoio.so.3.2
#8  0x00007f9b7104eaae in cv::VideoCapture::VideoCapture(cv::String const&) () at /usr/lib/libopencv_videoio.so.3.2
#9  0x00007f9b76436e32 in pyopencv_cv_VideoCapture_VideoCapture(_object*, _object*, _object*) () at /usr/lib/python2.7/site-packages/cv2.so
#10 0x00007f9b79d8dc60 in PyEval_EvalFrameEx () at /usr/lib/libpython2.7.so.1.0
#11 0x00007f9b79d918dc in PyEval_EvalCodeEx () at /usr/lib/libpython2.7.so.1.0
#12 0x00007f9b79d161bb in function_call () at /usr/lib/libpython2.7.so.1.0
#13 0x00007f9b79cef682 in PyObject_Call () at /usr/lib/libpython2.7.so.1.0
#14 0x00007f9b79cff05a in instancemethod_call () at /usr/lib/libpython2.7.so.1.0
#15 0x00007f9b79cef682 in PyObject_Call () at /usr/lib/libpython2.7.so.1.0
#16 0x00007f9b79d4c806 in slot_tp_init () at /usr/lib/libpython2.7.so.1.0
#17 0x00007f9b79d4905d in type_call () at /usr/lib/libpython2.7.so.1.0
#18 0x00007f9b79cef682 in PyObject_Call () at /usr/lib/libpython2.7.so.1.0
#19 0x00007f9b79d8d503 in PyEval_EvalFrameEx () at /usr/lib/libpython2.7.so.1.0
#20 0x00007f9b79d918dc in PyEval_EvalCodeEx () at /usr/lib/libpython2.7.so.1.0
#21 0x00007f9b79d919e8 in PyEval_EvalCode () at /usr/lib/libpython2.7.so.1.0
#22 0x00007f9b79dabefe in run_mod () at /usr/lib/libpython2.7.so.1.0
#23 0x00007f9b79dad1c1 in PyRun_FileExFlags () at /usr/lib/libpython2.7.so.1.0
#24 0x00007f9b79dae4e4 in PyRun_SimpleFileExFlags () at /usr/lib/libpython2.7.so.1.0
#25 0x00007f9b79dc0ca0 in Py_Main () at /usr/lib/libpython2.7.so.1.0
#26 0x00007f9b796fb43a in __libc_start_main () at /usr/lib/libc.so.6
#27 0x0000560a7bbd17ea in _start ()
```

FFMPEG probe:
```
#0  0x00007f9b79a9389d in read () at /usr/lib/libpthread.so.0
#1  0x00007f9b697e9ac8 in read (__nbytes=<optimized out>, __buf=<optimized out>, __fd=<optimized out>) at /usr/include/bits/unistd.h:44
#2  0x00007f9b697e9ac8 in file_read (h=<optimized out>, buf=<optimized out>, size=<optimized out>) at libavformat/file.c:112
#3  0x00007f9b697ca705 in retry_transfer_wrapper (transfer_func=0x7f9b697e9ab0 <file_read>, size_min=1, size=161649, buf=0x560a7e1afd3f "", h=0x560a7dd2a600) at libavformat/avio.c:378
#4  0x00007f9b697ca705 in ffurl_read (h=0x560a7dd2a600, buf=0x560a7e1afd3f "", size=161649) at libavformat/avio.c:411
#5  0x00007f9b697cd07a in avio_read (s=s@entry=0x560a7e0dbf40, buf=0x560a7e1afd3f "", size=161649, size@entry=262144) at libavformat/aviobuf.c:625
#6  0x00007f9b697f2535 in av_probe_input_buffer2 (pb=0x560a7e0dbf40, fmt=0x560a7e14ba28, filename=filename@entry=0x560a7e09c6c4 "/dev/video0", logctx=logctx@entry=0x560a7e14ba20, offset=offset@entry=0, max_probe_size=1048576) at libavformat/format.c:314
#7  0x00007f9b698fe9b2 in init_input (options=0x7ffd5a574dc0, filename=0x560a7e09c6c4 "/dev/video0", s=0x560a7e14ba20) at libavformat/utils.c:421
#8  0x00007f9b698fe9b2 in avformat_open_input (ps=0x560a7e14b710, filename=0x560a7e09c6c4 "/dev/video0", fmt=<optimized out>, options=0x560a7e14b9c8) at libavformat/utils.c:536
#9  0x00007f9b7106c469 in CvCapture_FFMPEG::open(char const*) () at /usr/lib/libopencv_videoio.so.3.2
#10 0x00007f9b7106c6df in cvCreateFileCapture_FFMPEG () at /usr/lib/libopencv_videoio.so.3.2
#11 0x00007f9b71070a13 in cvCreateFileCapture_FFMPEG_proxy(char const*) () at /usr/lib/libopencv_videoio.so.3.2
#12 0x00007f9b7104d45d in cvCreateFileCaptureWithPreference () at /usr/lib/libopencv_videoio.so.3.2
#13 0x00007f9b7104dd3c in cv::VideoCapture::open(cv::String const&, int) () at /usr/lib/libopencv_videoio.so.3.2
#14 0x00007f9b7104eaae in cv::VideoCapture::VideoCapture(cv::String const&) () at /usr/lib/libopencv_videoio.so.3.2
#15 0x00007f9b76436e32 in pyopencv_cv_VideoCapture_VideoCapture(_object*, _object*, _object*) () at /usr/lib/python2.7/site-packages/cv2.so
#16 0x00007f9b79d8dc60 in PyEval_EvalFrameEx () at /usr/lib/libpython2.7.so.1.0
#17 0x00007f9b79d918dc in PyEval_EvalCodeEx () at /usr/lib/libpython2.7.so.1.0
#18 0x00007f9b79d161bb in function_call () at /usr/lib/libpython2.7.so.1.0
[... cut ...]
#26 0x00007f9b79d918dc in PyEval_EvalCodeEx () at /usr/lib/libpython2.7.so.1.0
#27 0x00007f9b79d919e8 in PyEval_EvalCode () at /usr/lib/libpython2.7.so.1.0
#28 0x00007f9b79dabefe in run_mod () at /usr/lib/libpython2.7.so.1.0
#29 0x00007f9b79dad1c1 in PyRun_FileExFlags () at /usr/lib/libpython2.7.so.1.0
#30 0x00007f9b79dae4e4 in PyRun_SimpleFileExFlags () at /usr/lib/libpython2.7.so.1.0
#31 0x00007f9b79dc0ca0 in Py_Main () at /usr/lib/libpython2.7.so.1.0
#32 0x00007f9b796fb43a in __libc_start_main () at /usr/lib/libc.so.6
#33 0x0000560a7bbd17ea in _start ()
```

bgottula, 2017-09-02:
Justin implemented an experimental replacement for the OpenCV video capture object. Brief testing shows that the new approach is able to open the Logitech camera much quicker. We briefly attempted to use it while performing alignment but ran into trouble due to excessive latency of returned video frames; see #32.

jgottula, 2017-09-04:
#32 has been resolved and the **v4l2_capture** branch has been merged into **master**. Closing.

closed 2017-09-04 by jgottula