seeing-things/track #14: Add logging feature

bgottula, 2017-08-06:
This should replace most if not all print statements remaining in the code.

bgottula, 2018-03-01:
An alternative that may ultimately be more useful is to dump status information into a time-series database like InfluxDB which could be queried later for post processing or in real-time via Grafana.

bgottula, 2018-03-01:
Grafana also seems to support an interface that bypasses the database part if that is desired: https://grafana.com/plugins/grafana-simple-json-datasource

jgottula, 2018-03-05:
For Grafana, [Chronix](http://chronix.io) might be a reasonable choice of database. (Depending on how easily we can shove data into it from Python.)

bgottula, 2018-03-05:
I did some experimentation with InfluxDB and Grafana. Both seem to be reasonable options. Some relevant info from my experimentation:

InfluxDB downloads: https://portal.influxdata.com/downloads
I can't figure out how to download older versions (for compat with the python client) so whatever. I downloaded the .deb and installed with dpkg. Then fired it up with `sudo influxdb start`.

Did this little experiment with the built-in CLI console:
```
rgottula@mars-linux ~/Desktop $ influx
Connected to http://localhost:8086 version 1.4.3
InfluxDB shell version: 1.4.3
> CREATE DATABASE mydb
> show databases
name: databases
name
----
telegraf
_internal
mydb
> use mydb
Using database mydb
> insert voltage value=0.1
> insert voltage value=0.2
> insert voltage value=0.3
> insert voltage value=0.4
> insert voltage value=0.5
> insert voltage value=0.6
> insert voltage value=0.7
> insert voltage value=0.8
> insert voltage value=0.9
> insert voltage value=1.0
> select "value" from "voltage"
name: voltage
time                value
----                -----
1520228244746375196 0.1
1520228246211355351 0.2
1520228247711404910 0.3
1520228248790258341 0.4
1520228249905242588 0.5
1520228250760218663 0.6
1520228251892225760 0.7
1520228252690161290 0.8
1520228253502259118 0.9
1520228256646871368 1
```

Next I installed Grafana. This was another case of download the .deb, install with dpkg. Then start up the service: `sudo service grafana-server start`

I was then able to pull up the web UI at http://localhost:3000. I had to log in as 'admin' with pwd 'admin'. There's probably some way to turn that off. I added the InfluxDB datasource which was pretty easy, and then added a graph panel. It took me a little bit of fussing to get the database query just right--eventually I switched to the direct text edit mode and made this query for my 'mydb' database that I created above:

`SELECT "value" FROM "voltage" WHERE $timeFilter`

With that I was able to see a plot of the ten values I added manually!

Next I moved on to using the Python client, python3-influxdb: http://influxdb-python.readthedocs.io/en/latest/
I installed this with pip3 to get a recent version. The docs indicate that it hasn't been tested with newer than InfluxDB 1.2.4 which is concerning, but I'm using it with 1.4 and haven't experienced any issues yet. I copied one of the most basic examples from the documentation and modified it as follows, which worked great:

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
"""Tutorial on using the InfluxDB client."""

import time
import datetime
import random
import argparse

from influxdb import InfluxDBClient

def make_value():
    json_body = [
        {
            "measurement": "voltage",
            "time": datetime.datetime.now(datetime.timezone.utc).isoformat(),
            "fields": {
                "value": random.random(),
            }
        }
    ]
    return json_body

def main(host='localhost', port=8086):
    """Instantiate a connection to the InfluxDB."""
    user = 'root'
    password = 'root'
    dbname = 'mydb'

    client = InfluxDBClient(host, port, user, password, dbname)

    while True:
        json_body = make_value()
        print("Write points: {0}".format(json_body))
        client.write_points(json_body)
        time.sleep(0.5)

def parse_args():
    """Parse the args."""
    parser = argparse.ArgumentParser(
        description='example code to play with InfluxDB')
    parser.add_argument('--host', type=str, required=False,
                        default='localhost',
                        help='hostname of InfluxDB http API')
    parser.add_argument('--port', type=int, required=False, default=8086,
                        help='port of InfluxDB http API')
    return parser.parse_args()

if __name__ == '__main__':
    args = parse_args()
    main(host=args.host, port=args.port)
```

Once this was running I could see the data in the same graph in the Grafana dashboard in Chrome.

bgottula, 2018-03-29:
Experimental/preliminary support for telemetry logging to an InfluxDB has been pushed to the "telemetry" branch.

bgottula, 2018-04-01:
Telemetry logging to InfluxDB has been added. A preliminary set of telemetry channels were added for the Tracker, BlindErrorSource, OpticalErrorSource, HybridErrorSource, and Gamepad classes. Basic testing with the executable programs was successful.

closed 2018-04-01 by bgottula