seeing-things/track #21: Support video output to file while tracking with camera

bgottula, 2017-08-12:


jgottula, 2017-09-04:
The newer fork of `python-v4l2capture` supports specifying a FOURCC code when calling `v4l2capture.Video_device.set_format`, and specifying `"JPEG"` does seem to cause `v4l2capture.Video_device.read` returning un-decoded JPEG data instead of pre-decoded 24-bit RGB data.

This should make it possible to do video output to Motion JPEG without re-compressing the frames in any way, which ought to give us both zero loss of quality and great performance.

bgottula, 2017-09-05:
Are you able to confirm that the library is in fact returning the raw JPEG frame from the camera when using the "JPEG" FOURCC code rather than decoding it and re-encoding it?

bgottula, 2018-03-31:
I still want this. :grimacing:

bgottula, 2018-05-23:
There is fairly good evidence from 21 May 2018 tracking telemetry that noise in the optical error source estimates are causing jerkiness in tracking performance. If raw webcam imagery from that pass were available it would have been possible to confirm this and test adjustments to the algorithm on realistic inputs.

bgottula, 2018-06-06:
Added support for dumping JPEG frames to files on disk. Code is largely copied from the dump_test branch.

I tried searching for ways to mux the frames into an AVI file in real-time in Python but could only find references to using OpenCV or ffmpeg. I decided not to go the OpenCV route because (if I'm not mistaken) this involves decoding the JPEG to a numpy array and then re-encoding it using the VideoWriter class. What I actually want is to create an AVI file with the exact same data that the camera is spewing out without re-encoding it. So I opted to revert to dumping the frames to disk, which can then be easily muxed into an AVI file using the jpg_to_avi.sh script in the utils/ directory.

If the kwarg to enable the frame dump is given, the program currently just creates a new numbered directory from wherever the program was run from. Perhaps it would be better to make this configurable but it's good enough for now. At least the current approach does not require hand-editing program arguments on each program invocation in order to avoid re-using the same directory.

closed 2018-06-06 by bgottula