seeing-things/track #29: Alignment: improve reliability of object apparent motion estimation

jgottula, 2017-08-14:


bgottula, 2017-09-02:
During alignment the object's apparent motion is derived from the tracking loop's slew rates after the loop has converged. However the accuracy of this approach requires that the tracking loop be fully converged, which may be difficult to detect and may take longer than desired. Instead, I think it would be better to center the object and then stop slewing the mount and simply observe the object's apparent motion in the camera with the mount stationary until the object comes near the edge of the frame or a certain amount of time (15-30 seconds) has elapsed.

bgottula, 2017-09-03:
Fix has been applied but not tested. Closing on assumption that this fixes the problem. Reopen if evidence surfaces that it is still a problem.

bgottula, 2017-09-03:
Apparent motion estimation is still a bit unreliable in the case where the object's apparent motion is very small in either axis. If the apparent motion is small compared to the jitter and other sources of noise in its estimated position, the estimate of its motion will be poor. Some ideas to improve the situation:

- Use all of the error estimates during the observation window rather than just the start and stop points. Should be possible to average out some of the noise to improve the accuracy of the estimate.
- Ensure that the object has moved by a statistically significant amount in both axes before moving on. If no motion is detected after some max time limit, give up and instruct the user to select a different star.

bgottula, 2017-09-03:
For all cases, the estimate of object motion needs to detect the direction with high confidence. For case b it must also estimate the rate of motion in each axis with reasonable confidence so that the algorithm can estimate how close to the center of frame the object must be placed such that it will drift past the center within a reasonable time interval. For case c the estimate of the direction angle must also be fairly accurate so the algorithm can accurately place the object in the frame such that its motion vector will intersect the center of the frame.

bgottula, 2017-09-03:
It may be appropriate to estimate the variance of the error estimates over some set of samples while the tracking loop is converged. This information can be used to determine how many samples need to be collected and/or how far the object needs to move to produce a reliable estimate. Code for this can be recovered from the deleted backlash estimation script.

bgottula, 2017-09-05:
I made some additional adjustments. After some further thought, I realized that the tracking loop is doing the exact kind of filtering on the error source samples that we want to apply. However, as observed previously, it is more reliable to take the difference in the mount's position after some time interval than to use the current slew rate for estimation of object apparent motion.

I added code to take statistics on the error source estimates however the stats are only used in print statements. If the tracking loop output is used it would be more appropriate to measure the variance of the mount position (after subtracting the expected motion) and use this information to determine if the position has changed by a significant enough amount to produce a reliable estimate of object motion (direction, angle, and magnitude).

bgottula, 2017-10-07:
This issue is obsolete due to change in alignment strategy--see #43.

closed 2017-10-07 by bgottula