seeing-things/track #37: Ensure correct versions of Python library dependencies

jgottula, 2017-09-04:
Unfortunately, python-v4l2capture turns out to be one of those libraries where the original "canonical" version hasn't been updated in the better part of a decade, and so there are various forks of the project to improve it, which of course have never been re-integrated back into the "trunk" repository (as the original author is apparently dead, or has otherwise fallen off the face of the earth, or just doesn't want to merge any improvements that other people has made to his software for some stupid reason).

Pypi hosts the original (old) version, whose source code is located on the Launchpad website (essentially a far less usable GitHub analogue, designed for the bizarre Bazaar source control management system, which in typical Canonical fashion is something that Canonical invented for the sake of being different and which essentially no one besides Canonical actually uses).
https://pypi.python.org/pypi/v4l2capture
https://launchpad.net/python-v4l2-capture

This is the main GitHub fork of the project, which has actually seen non-negligible amounts of development since 2011 (including FOURCC support so that you can actually capture the JPEG frames rather than being forced to auto-decode them to 24-bit RGB; getters and setters for the camera control parameters; and Python 3 support).
https://github.com/gebart/python-v4l2capture

Based on the things I want to do, we need to use the latter version (the gebart GitHub fork).

But as a result of the fragmentation, repository packages for the library may or may not be the version we want. And unfortunately forking doesn't tend to produce the best versioning continuity. So for example, on Arch Linux, where I work, the Python 2 package for the library is the old crusty version from Launchpad, while the Python 3 package for the library is the GitHub version (but not the gebart fork; it's actually *a fork of the gebart fork* which added Python 3 compatibility, which has actually already already been merged back into the gebart repo).

So, @bgottula, double-check and make sure you have the correct version on Mint; and then we probably ought to add a README file documenting which fork of the library is supposed to be used with our software.

There's also a similar situation going on with our usage of PIL, the Python Imaging Library. The original version of that library hasn't been updated since 2009, and the modern fork that everyone (including us) uses is Pillow, which has been actively developed and changed and has Python 3 support. But it's still imported with the `PIL` name, despite the API having been changed in various ways. So we ought to document that as well.

jgottula, 2017-10-28:
The need for v4l2capture at all may go away, since I've already begun interfacing with the webcam manually for some things (e.g. ioctls for setting camera parameters) and will very likely end up manually implementing the few things that I currently use v4l2capture for.

PIL will remain a dependency.

bgottula, 2017-11-05:
The v4l2capture dependency is now handled by means of an entry in the dependency_links kwarg in setup.py such that the gebart fork on Github will be downloaded and installed rather than the decrepit version on PyPI.

closed 2017-11-05 by bgottula