seeing-things/track #40: Automatically set webcam controls from Python

jgottula, 2017-09-04:
Instead of requiring the user to remember to do separate webcam setup via the `v4l2-ctl` CLI utility, set the controls to the desired values within the Python scripts themselves, preferably via `v4l2capture`, or via `v4l2` when necessary.

Most important controls:
- [ ] Automatic gain: always disable
- [ ] JPEG compression quality: always max allowed
- [ ] Exposure: optional cmdline param with reasonable default value?

Less important controls:
- [ ] Brightness
- [ ] Contrast
- [ ] Gamma
- [ ] Sharpness
- [ ] Power line frequency

jgottula, 2017-09-05:
Implemented the 3 higher-priority tasks.

bgottula, 2018-03-31:
The default values for the remaining controls seem to work just fine. Cookie Monster criterion achieved! :cookie:

closed 2018-03-31 by bgottula