seeing-things/track #41: Auto-determine a sane value for camera bufs based on camera FPS, if possible

jgottula, 2017-09-05:


jgottula, 2017-10-28:
Might still need to check whether there's a useful v4l2 ioctl for this that I can directly query; previously I may only have been looking at what the v4l2capture module exposes.

jgottula, 2018-04-02:
This needs a bump.

Basically if we want to be able to withstand a 1-second hangup on consumption of frames by the tracker, and we're hypothetically capturing at 15 fps, then we need 15 buffers to satisfy that 1-second worst-case condition.

The problem right now is that, after looking into the available IOCTLs from V4L2, and implementing both `VIDIOC_ENUM_FRAMEINTERVALS` and `VIDIOC_G_PARM` (each of which should be able to give framerate information), our webcam and/or driver seems to be incapable of returning any such information.

I really want to get rid of that obnoxious mandatory `--camera-bufs` command line argument though, so I may just make it optional and set an arbitrary default that's, like, really high. If it's too high, then the `VIDIOC_REQUESTBUFS` IOCTL will just give us fewer than we asked for. And we can have the default be like 100 or whatever, it doesn't really matter as long as it's a reasonably high figure, since we seem to be incapable of basing it on Real Information such as the camera's framerate.

bgottula, 2020-01-02:
Webcam has been retired from our hardware lineup so this is no longer a priority.

closed 2020-01-02 by bgottula