seeing-things/track #42: Assisted alignment case B problem

jgottula, 2017-09-05:
Case B of alignment failed to drop off object within 30 seconds of it crossing the center of the FOV.

This came up while aligning on Saturn tonight.

bgottula, 2017-09-06:
Each time we had a Case B alignment the Alt axis was moving opposite of the alignment direction. The algorithm moved Saturn to the top edge of the frame, slew past the backlash deadband (which didn't cause Saturn to move much, as expected) but then it immediately stopped slewing even though Saturn was a few minutes away from crossing frame center. I noted on one instance that the estimated rate of motion in Altitude was totally bogus--something like 200-300 arcseconds/second, so apparently #29 has not been fully resolved.

bgottula, 2017-09-25:
This issue is obsolete due to change in alignment strategy--see #43.

closed 2017-09-25 by bgottula