seeing-things/track #47: "File too large" error when opening camera

bgottula, 2017-10-02:
The new WebCam class will occasionally fail to open the camera with an error message that includes the statement "file too large." I don't have the full error message handy but I can try to reproduce it if necessary.

jgottula, 2017-10-16:
"File too large" corresponds to errno `EFBIG`.

It's reasonably likely that this is coming from the general Linux kernel USB code (something about too many ISO frames); and if that's the case, it appears to be the sort of error that is USB-host-controller-specific (so it may only occur on particular USB ports of particular computers).

Also this is most likely happening in the relatively recently added code in the `WebCam` class for setting the `v4l2` control values (either in the device `open` or maybe in one of the `ioctl` calls).

Will try to get more info the next time this happens. I don't believe I've ever encountered this error myself; and it's plausible that it might have only happened while Brett was at his main desktop computer and not the MacBook (different USB host controller).

bgottula, 2017-10-16:
```
rgottula@hyperion ~/dev/track $ ./hybrid_track.py --max-divergence=5.0 tle ./what.tle 
In TLE file mode: './what.tle'.
libv4l2: error turning on stream: File too large
Traceback (most recent call last):
  File "./hybrid_track.py", line 92, in <module>
    args.max_divergence
  File "/home/rgottula/dev/track/errorsources.py", line 360, in __init__
    self.optical = OpticalErrorSource(cam_dev_path, arcsecs_per_pixel, cam_num_buffers, cam_ctlval_exposure)
  File "/home/rgottula/dev/track/errorsources.py", line 258, in __init__
    self.webcam = webcam.WebCam(cam_dev_path, cam_num_buffers, cam_ctlval_exposure)
  File "/home/rgottula/dev/track/webcam.py", line 62, in __init__
    self.camera.start()
IOError: [Errno 27] File too large
```

bgottula, 2017-10-28:
If there isn't an easy way to fix the root cause of the problem, consider catching the error/exception and retrying a few times before giving up.

jgottula, 2017-10-28:
I wasn't able to pin down the source of the `EFBIG`, so yeah, the most reasonable thing to do here is probably to explicitly catch cases of `EFBIG` and retry.

bgottula, 2018-05-04:
I haven't seen this in quite a while so I'm going to close it for now. If I see it again I'll reopen.

bgottula, 2018-05-09:
Got this error again today.

```
rgottula@hyperion ~/dev/track $ ./track/align.py --telem-enable
Traceback (most recent call last):
  File "./track/align.py", line 339, in <module>
    main()
  File "./track/align.py", line 235, in main
    meridian_side=args.meridian_side
  File "/home/rgottula/dev/track/track/errorsources.py", line 479, in __init__
    mount=mount
  File "/home/rgottula/dev/track/track/errorsources.py", line 223, in __init__
    self.webcam = webcam.WebCam(cam_dev_path, cam_num_buffers, cam_ctlval_exposure)
  File "/home/rgottula/dev/track/track/webcam.py", line 39, in __init__
    self.start()
  File "/home/rgottula/dev/track/track/webcam.py", line 88, in start
    self._v4l2_ioctl(v4l2.VIDIOC_STREAMON, ctypes.c_int(int(v4l2.V4L2_BUF_TYPE_VIDEO_CAPTURE)))
  File "/home/rgottula/dev/track/track/webcam.py", line 236, in _v4l2_ioctl
    assert (fcntl.ioctl(self.dev_fd, req, arg) == 0)
```

bgottula, 2018-08-26:
Well, once again it has been several months and I have not seen this error again since. Closing again, hoping it stays away more than five days this time.

closed 2018-08-26 by bgottula