seeing-things/track #58: New multiprocess camera code prevents clean CTRL-C shutdown

bgottula, 2017-10-21:
During the first night of observing with the new multiprocess camera code (on 10/15) I had difficulty shutting down the program cleanly. I had to use CTRL-C twice before the program fully quit, and subsequently the camera was unavailable to programs attempting to access it. This forced me to unplug the camera and plug it in again, which then resulted in the camera showing up under a different name in the filesystem, requiring me to make edits to the configuration file. This was disruptive during our busy observing session so we ended up reverting these commits on the master branch.

Please continue work on the webcam_concurrency feature branch (rebase it on master if desired) and test shutdown cases carefully, noting results in comments.

jgottula, 2017-10-21:
@bgottula Okay so I've rebased to master (without the revert commits) and done a few things:

- From some research I did, apparently Pythom multiprocess actually makes *all* processes get the ctrl-C `KeyboardInterrupt` exception. (Or so I'm led to believe by Internet people.) So there's now a handler for `KeyboardInterrupt` in the webcam process monitor loop; the handler causes that method to return and therefore terminate the process. (It will print a message if it catches that exception, so that you'll know whether or not it actually happens.)
- I reorganized the webcam shutdown code a bit, and more importantly added a bunch of debug printouts to the `WebCam` dtor so that you'll know which things it's doing as it attempts to stop the monitor process; and also if it gets stuck somewhere in the shutdown process, you'll know what it's gotten stuck on.
- I also added a statement in the monitor process initialization that sets the process's `daemon` attribute to `True`. [According to the docs](https://docs.python.org/2/library/multiprocessing.html#multiprocessing.Process.daemon), setting that flag should cause the main process to terminate the child process when the main process exits.
  - A note on this: the way the code is currently in the branch means that the `WebCam` dtor will try to do `Process.terminate()` (effectively `SIGTERM`) as a last resort, and by definition that must necessarily occur before the main process actually finishes executing; and [allegedly](https://docs.python.org/2/library/multiprocessing.html#multiprocessing.Process.terminate) doing `Process.terminate()` can cause some shared resources to not be cleaned up properly (perhaps leaving the webcam device node open? though that would contradicts everything I know about how process termination and resource management works in Linux/POSIX). So it might be worth trying commenting out the `terminate()` part of `WebCam.stop_monitor_proc` to see if perhaps skipping that step (and then letting the main process exit, ostensibly terminating its daemonic child processes) works any better or worse.

I'm kinda hoping here that the improved diagnostic printouts will mean that this is good enough for you to test yourself; I don't think I was ever necessarily clear on what the exact circumstances *were* that led to the exit hang-up, and your more full setup may be more amenable to reproducing those circumstances.

But let me know if you want me to help test.

bgottula, 2017-10-21:
Awesome, thanks! I will test it in the next hour or so.

bgottula, 2017-10-21:
When running optical_track.py, CTRL-C prints a message but the main process keeps on running. Further CTRL-C's have no effect. The OpenCV window remains open and the mount continues slewing (the dtor for the NexStar class sends a command to stop slewing).

```
rgottula@mars-linux ~/dev/track $ ./optical_track.py 
No gamepads found.
^CWebcam process caught KeyboardInterrupt
^C^C^C^C^C^C^C^C^C^C^C^C^C^C
^Z
[1]+  Stopped                 ./optical_track.py
```

jgottula, 2017-10-22:
@bgottula Okay I realized that the core of the problem here is that the main thread is still blocking in `WebCam.get_fresh_frame` (`poll`ing on the pipe thru which frames are sent between the processes).

I've pushed a couple of commits that *should* cause the monitor process to close that pipe when it terminates, which I believe will cause the main process in `WebCam.get_fresh_frame` to get an `EOFError`, or something. Let me know if that happens now.

If it does throw `EOFError` now, then that's good; I'll still need to do a little bit of stuff afterwards to ensure that the exception is caught and that the function returns in a clean/sane manner.

(I sort of forgot that unlike in regular threading models, Python's idiotic `multiprocessing` paradigm puts the child "thread" into a separate process, which means that all of the file descriptors and such are *duplicated* at the moment of "thread" creation, not *shared* between threads as they would be in a sane/standard, actually-uses-threads multithreaded program. It just makes everything that much more complicated and horrible...)

jgottula, 2017-10-22:
And if this doesn't work, then I'll just do some slightly-less elegant stuff in `get_fresh_frame` involving timeouts and periodic checking, rather than sitting in a single blocking syscall until something we care about happens; the latter case is really the ideal way to do it but if it isn't easily workable for what we want, then screw it.

bgottula, 2017-10-22:
Well I don't know for certain if it threw EOFError, but it did appear to shut down cleanly. I was also able to re-run the program without any camera access issues.

```
WebCam: get_fresh_frame poll end
WebCam: get_fresh_frame poll begin
WebCam: get_fresh_frame poll end
WebCam: get_fresh_frame poll begin
WebCam: get_fresh_frame poll end
WebCam: get_fresh_frame poll begin
WebCam: get_fresh_frame poll end
WebCam: get_fresh_frame poll begin
WebCam: get_fresh_frame poll end
WebCam: get_fresh_frame poll begin
WebCam: get_fresh_frame poll end
^CGoodbye!
rgottula@mars-linux ~/dev/track $ 
```

bgottula, 2017-10-22:
Also noticed that the colors of the camera frame are different compared to master branch and the latency is much worse. Does this branch not set the camera parameters like exposure?

jgottula, 2017-10-22:
> Well I don't know for certain if it threw EOFError, but it did appear to shut down cleanly. I was also able to re-run the program without any camera access issues.

Okay, well, that sounds provisionally good then; I'll assume for now that the shutdown problem is essentially resolved.

> Also noticed that the colors of the camera frame are different compared to master branch and the latency is much worse. Does this branch not set the camera parameters like exposure?

That's really weird; there should be no differences. Is it possible to check with `v4l2-ctl` in each case to compare?

Now that I think about it, I wonder if this whole "spawning additional process means that all of the file descriptors and whatnot are duplicated" screws stuff like that up. In the `WebCam` ctor, I open the webcam device node, call the ioctls to set the parameters, and then after that I spawn the separate monitor process; and I don't close the file descriptor for the webcam device before spawning that process.

This whole pile of 💩 almost makes me want to redo this whole thing from scratch using the threading methodology, because GIL contention is preferable to file descriptor duplication and head-scratching resource management hell. And if I were to implement CPU-intensive transcoding or something, I'd probably move just that CPU-bound kind of stuff to a child process, and leave the device access stuff in a thread for simplicity's sake.

Ironically, programming this kind of threaded stuff in C or C++ would be more straightforward than doing it in Python (at least for me). That's a sad, sad state of affairs for Python...

bgottula, 2017-10-23:
The gain_automatic and exposure settings are as expected according to the v42l-ctl utility. But the latency is quite noticeably worse. You should be able to test this yourself--just replace `tracker.run()` in optical_track.py with a loop that calls `error_source.compute_error()` repeatedly and compare between the two branches.

I'll leave the path forward up to you, but I would definitely prefer to err on the side of "do the simplest thing possible that's good enough" for the sake of expediency and future maintainability. In fact I wouldn't mind running everything single-threaded like on master if the latency penalty is small even after adding video output. At the end of the day the things I care about roughly in order are (1) latency, (2) ease of use in the field (which includes things like graceful shutdown), (3) maintainable code that's not hard to understand, and (4) capturing video to a file. I'm not opposed to multiprocessing or threads but they do tend to create extra problems like this that take longer to debug and really understand. And like you said, threading/multiprocess in Python comes with some bizarre side-effects.

jgottula, 2017-10-28:
Given the difficulties and problems associated with this, for now I think we'll just leave master on the non-parallel implementation and leave the webcam_concurrency branch around in case it ever becomes really actually truly important that we have parallel webcam stuff (enough to justify dealing with Python's horrendousness).

closed 2017-10-28 by jgottula