seeing-things/track #111: optical_track shutdown error

bgottula, 2019-01-20:
Got the following error from optical_track. I think this occurred after I used CTRL-C to shut it down.

```
rgottula@nix ~ $ optical_track 
Gamepad found and registered.
^CGoodbye!
^CException ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```

jgottula, 2019-08-04:
We did a bunch of debugging on this today to figure out what's going on. In the cases where we reproduced the issue, there were 2 threads still running: the main thread (which was using ~20% CPU and appeared to be actively issuing Gemini2 commands over the UDP socket), and another thread (which was not using any CPU, and had a call stack that indicated that it was probably a gamepad-related USB thread, waiting or possibly stuck in `poll`). Those stack traces are in the Slack DM's.

One thing worth revisiting would be the per-thread `strace` logs idea, which would help determine exactly which fd was being `poll`d upon by which thread and for how long, etc.

Also we might want to look into updating the Python inputs library from 0.1 (what we are using) to 0.5 (the latest), as it appears that a bunch of relatively recent development has gone into that, which may or may not help matters (as the gamepad code is one of the suspects here).

jgottula, 2019-08-04:
Actually it's not necessarily a mystery as to why Gemini2 stuff was still happening in the main thread; the destructor for the `Gemini2` class calls the `stop_motion` method, which itself contains a `while True` loop that *could potentially* have problems if erroneous conditions cause it to never terminate. So that's worth investigating.

bgottula, 2019-08-04:
We also discussed some options for ensuring that the mount motion stops even if other threads misbehave. The most promising idea is to explicitly call the destructor on the Mount object in the exception handler for `KeyboardInterrupt` so that it is called promptly, because otherwise it is not called until all other threads stop and the main thread finishes its cleanup processing (which may never happen if the process has to be terminated in some other way).

It may also be worth investigating adding handlers for other signals like SIGTERM.

bgottula, 2019-08-05:
Okay. I think we may have solved the most pressing issue, which is that mount safing will occur without first waiting for the other threads to end. I'm going to close, but if we continue to encounter issues this could be reopened. We didn't really identify root cause or correct the problems preventing the other threads from ending.

closed 2019-08-05 by bgottula