seeing-things/track #131: Get location from GPS when available

bgottula, 2019-11-10:
Add support for querying gpsd for location information obtained from a GPS receiver. This functionality should allow, at minimum:

- Retrieval of latitude, longitude, and altitude
- Retrieval of GPS receiver status and solution quality and confirmation that it meets some simple criteria. Examples: Does it have a fix? Is the error on the position estimate below some threshold?

I can think of a couple ways to integrate this into the main application code:
- Always query gpsd at startup. If the GPS solution is deemed to be good, use it, even if location info is provided as program arguments. If not, print a loud warning and fall back on the program arguments.
- Use argument-supplied location information when provided in config file or on command line. Only query GPS when this is not provided. If GPS solution is bad, abort the program with a failure message.

Rationale: Observer location is currently given in a config file or on the command line. This works okay but it is easy to forget to update the location. This can lead to poor alignment and inaccurate pointing.

jgottula, 2019-11-10:
We're now going to only query the GPS during alignment. It'll be a one-time thing, at the start of the alignment process, where we block for a period of time and get a GPS location, and then store that along with the other alignment crap in some serialized format that then goes on to get used by other stuff.

The way I'll implement this is to write a function that basically is "get me the lat/lon/alt please", which is blocking and which will keep reading the GPS until it gets a fix *and* is satisfied that it is a good fix. (Or until a timeout is reached, at which point it should most likely terminate with an error. Something like that.) This function will use the gpsd code that allows for continuously watching the gps device output via JSON responses that are automatically converted into dict-like objects. It will connect to the gps device; wait until it has a fix; wait until that fix is "suitably good" (TBD), and then disconnect from the gps device and return The Location which will be used for the alignment and all future observation stuff based on that alignment.

jgottula, 2019-11-10:
http://manpages.ubuntu.com/manpages/bionic/man5/gpsd_json.5.html

bgottula, 2020-04-18:
This feature has been implemented and some basic tests have passed. Programs that rely on a mount model will always get observer location via that model rather than from GPS or program arguments so that the observer location is consistent with the location used during alignment. For the align program and programs that do not use a mount model the location can either come from GPS if no program args are provided or if the lat, lon, and elevation program args are all provided they override GPS. It is an error when some but not all of lat, lon, and elevation are provided.

closed 2020-04-18 by bgottula