seeing-things/track #138: Try an autoformatter

bgottula, 2019-11-11:
Try using an autoformatter on this codebase, either as a one-time thing or (ideally) run as part of some kind of pre-commit hook and/or in CI that must pass before a PR can be merged. The popular ones seem to be autopep8, yapf, and black. I don't know a ton about any of them, but their proponents argue that autoformatters are good because:

- Less time spent on code formatting = more time spent writing code
- Less time spent debating or discussing code formatting
- Guarantees consistent code style across the project

Black takes this a step further by offering very few configuration options such that you can't even argue about how to configure it.

jgottula, 2019-11-11:
This scares me. :fearful: 

For one thing, applying stuff like this to existing code destroys the ability to diff/blame things sensibly for no good reason. :frowning_face: 

And another reason I don't like these things, is that I will sometimes do things (in both C++ and Python) that may be "technically bad" but which make my code far more readable than it would otherwise be. When it makes sense, I will do stuff like putting `if` statements on individual lines, particularly when there are multiple such statements, one after another, that are doing similar things and which line up nicely if they are done as single-lines. And I like to manually space-pad out the RHS of groups of expressions (or the corresponding words/args of related lines of code that are making the same function call with different values), so that everything lines up super cleanly no matter how different the name lengths are or whatever.

It really bothers me when I attempt to read people's code that has clearly been autoformatted in such a way that it technically follows some rule to the letter, yet is made **less** readable as a result. One of the best recent examples of this that I can think of is the OpenRCT2 code, where they have `clang-format` enforce (among other things) line wrapping. And so you get tons of these places where the formatter has just mindlessly wrapped every line at whichever word it is where the line would pass column 80 or whatever, and so you get line breaks in places that aren't necessarily sensible (e.g. rather than putting the line break in a semantically reasonable place, it'll be right near column 80 and leave just one more tiny function argument on the next line or something stupid like that); and you also get a mishmash of nearby lines that are actually doing the same thing as each other, but some of them will be line-broken purely because they had longer-named things in them. Makes it a nightmare to clearly tell what I'm even looking at.

I'd be alright with a formatting tool (or config thereof) that enforces certain important things that we care about but isn't super forceful about inflicting its will onto every aspect of the code.

bgottula, 2019-11-11:
One of the main motivations is that I want the ability to run a linter in CI (e.g., pylint or pyflakes) and not have to disable 90% of the checks to get them to pass. We already don't have unit tests but linting will at least prevent facepalm-worthy mistakes like the missing imports that got merged to point.

I don't think the changes to the existing code in this repo will be drastic, though admittedly it will probably affect some of your stylistic choices more than mine. In fact I wanted to try it here first because I think the changes to the gemini source files in point will be more drastic.

Regarding blame: Not to worry, I will not forget who to blame, no matter what the tools determine. :smiling_imp:

jgottula, 2019-11-11:
> Regarding blame: Not to worry, I will not forget who to blame, no matter what the tools determine. :smiling_imp:

It's more about which commit than who did the change, but I get your point. :smile: 

> admittedly it will probably affect some of your stylistic choices more than mine

:angry: 

What if I can come up with a `.pylintrc` config that has a relatively small set of disablements that are sufficient enough to alleviate the majority of my grumpiness?

> linting will at least prevent facepalm-worthy mistakes like the missing imports that got merged to point

I'm totally with you on `pylint` being a good thing and something that we should turn into a necessary check.

Just yesterday I wrote a whole new Python program for personal use: a cron job script for the home server, to get the current wan-side IPv4 and IPv6 addresses and then update my dynamic DNS with the Google Cloud DNS API. For a very long time, I used ZoneEdit: they were frankly a really nice DNS provider for a long time but sadly they don't really get updated anymore. And while ZoneEdit had a super simple "just run `curl` at this URL and it'll automagically update your `A` record to the IP address the request came from" dynamic DNS update procedure, Google Cloud DNS is much more... "enterprisey", and so I had to write my own script to invoke their complicated API's (via a separate Node.js program that I can probably eliminate at some point :roll_eyes:).

Anyway, the point there was that when I thought I was "done" writing that script, I decided to run `pylint` on it, and I discovered a number of subtle errors in it. It's nice to have some level of static analysis so that you don't only ever discover problems literally at runtime when you happen to hit the exact right code path.

bgottula, 2022-12-05:
I gave Black a try today with the default settings (not that you can change much). It turns out that the way I format Python code is already apparently pretty darn close to what Black does, since it made fewer changes than I was expecting. A large number of the changes were replacing single quote characters `'` for double quote characters `"` around strings and re-wrapping of lines to the 88-character default line length limit since I've been using a 100 character limit. Both of those could be configured to preserve what I have now. Other common changes were using a trailing comma consistently at the end of lists, using spaces around math operators, and more consistent handling of how long lines are split over multiple lines. Fortunately I think it does reasonable things when splitting long lines; for example if a function call has five arguments and four would fit on one line but the fifth doesn't, it doesn't stupidly put just the fifth argument on a new line by itself; instead it puts each argument on its own line. Only if all arguments fit on one line does it put multiple arguments together on a line.

The TL;DR is that I was pleased with what I saw. I'll test drive it some more before I commit any wholesale formatting changes. Expect another update around Q4 of 2025.

bgottula, 2023-02-20:
With regard to blame, it's actually possible since Git 2.23 (released August 2019) to tell Git to ignore specific commits when assigning blame: https://black.readthedocs.io/en/stable/guides/introducing_black_to_your_project.html