seeing-things/track #143: Gamepad: Investigate ditching the inputs library and replacing with SDL

jgottula, 2019-11-28:
May fix #71.

Also likely to alleviate issues we’ve had with threads inside the inputs library getting stuck on quit, and other grievances with the inputs library being janky in general.

jgottula, 2019-11-29:
Endorsed SDL2 bindings are listed here: https://www.libsdl.org/languages.php

Also, I might consider just using the Python bindings to libudev and libevdev. SDL has the advantage of being very broadly cross-platform, and this approach would by contrast be effectively Linux-only. But I don’t know for sure how much of SDL’s functionality we’ll necessarily need. Might just be simpler to go low-level.

Then again, it might turn out that SDL offers better asynchronous operations and generally nicer abstractions to work with... will have to take a look to determine that.

bgottula, 2020-03-23:
I did some preliminary reading of the documentation for the python-evdev package and it looks like it could serve as a near drop-in replacement for inputs, but with a couple of key benefits:

- Non-blocking `read()` using `select()`
- Good documentation
- Built-in support for `asyncio`
- Repository is more active, possibly more widely used?
- Code is broken up into reasonable-sized modules

The level of abstraction appears to be similar to what is provided by the inputs package.

bgottula, 2020-06-14:
Deferring this for now. inputs may not be perfect but it's good enough. If the shutdown issues continue to be a problem we can re-explore this. At least one instance of needing to do CTRL-C more than once was the gamepad.py main() method, and the problem in that case was that the program did not call the `stop()` method on the Gamepad object and thus the threads did not get notified to shut down properly. This has since been fixed.

closed 2020-06-14 by bgottula