seeing-things/track #199: Need to CTRL-C twice to end program with mount multiprocessing

bgottula, 2020-12-06:
Not sure if the problem is in this project or in point, but I frequently need to CTRL-C twice to end the program (but not always). When I need to do it twice I get the following:

```
(venv) rgottula@nix:~/dev/track$ track --meridian-side=east --fuse solarsystem Mars
NTP check failed: PPS is not the selected NTP system peer
Continue anyway? [Y/n]: 
In named solar system body mode: 'Mars'
Sensor fusion with camera enabled
Could not connect to laser pointer FTDI device.
Gamepad found and registered.
^CGot CTRL-C, shutting down...
Safing mount...Mount safed successfully!
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 28, in poll
    pid, sts = os.waitpid(self.pid, flag)
KeyboardInterrupt
```
We've encountered numerous versions of this problem over the years, so check the issue history for some insights.

bgottula, 2021-01-17:
Today I've been having this problem, but it's worse than usual because nearly every time I have to end the program with a second CTRL-C, the two forked processes continue running and I have to hunt them down and kill them manually. Not sure what is different because I don't think I have typically needed to do this extra step.

jgottula, 2021-01-17:
I should be able to help with this.

I have lots of experience in the area of planning and executing the premeditated murder of CPU processes that have overstayed their welcome. 😉

Give me a call whenever, and I'll see what I can come up with.

bgottula, 2021-01-19:
Another variant, with multiprocessing disabled:

```
^CGot CTRL-C, shutting down...
Safing mount...stop motion called
Mount safed successfully!
^CException ignored in: <module 'threading' from '/usr/lib/python3.6/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 1294, in _shutdown
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
In Gemini2 destructor
stop motion called
returning from Gemini2 destructor
```

bgottula, 2021-01-19:
With multiprocessing enabled:

```
^CGot CTRL-C, shutting down...
Safing mount...stop motion called
called slew() with rate 0 on both axes, waiting on events
RA axis event is set
DEC axis event is set
returning from stop_motion()
Mount safed successfully!
^CException ignored in: <module 'threading' from '/usr/lib/python3.6/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 1294, in _shutdown
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 28, in poll
    pid, sts = os.waitpid(self.pid, flag)
KeyboardInterrupt
In Gemini2 destructor
In Gemini2 destructor: ra process is alive, sending None to pipe
In Gemini2 destructor: dec process is alive, sending None to pipe
rate_target is None for dec, setting div_target to 0 and shutdown to True
rate_target is None for ra, setting div_target to 0 and shutdown to True
shutdown == True for ra
div_last_commanded == 0, returning from ra
Finished sending None to both axis processes
Joining process for ra axis
shutdown == True for dec
div_last_commanded == 0, returning from dec
Join returned for ra
Joining process for dec axis
Join returned for dec
returning from Gemini2 destructor
```

bgottula, 2021-01-19:
With lots of print statements sprinkled around:

```
^CGot CTRL-C, shutting down...
Safing mount...stop motion called
called slew() with rate 0 on both axes, waiting on events
RA axis event is set
DEC axis event is set
returning from stop_motion()
Mount safed successfully!
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 28, in poll
    pid, sts = os.waitpid(self.pid, flag)
KeyboardInterrupt
In Gemini2 destructor
In Gemini2 destructor: ra process is alive, sending None to pipe
In Gemini2 destructor: dec process is alive, sending None to pipe
Finished sending None to both axis processes
Joining process for ra axis
rate_target is None for ra, setting div_target to 0 and shutdown to True
rate_target is None for dec, setting div_target to 0 and shutdown to True
shutdown == True for ra
div_last_commanded == 0, returning from ra
shutdown == True for dec
div_last_commanded == 0, returning from dec
Join returned for ra
Joining process for dec axis
Join returned for dec
returning from Gemini2 destructor
```

bgottula, 2021-01-30:
Had a call with @jgottula on 18 Jan 2021 where we discussed this in depth. Some of the main conclusions we came to are:

- The way multiprocessing works in the point package right now is kinda scary, and surprising that it works at all.
- Ultimately, it would be much better if we could rely on Gemini to handle acceleration asynchronously so that we don't need to manage this and resort to using multiprocessing or other fancy and complicated things. See https://github.com/seeing-things/point/issues/46 for more on that.

I think it would be best to pursue feasibility of ditching multiprocessing before investing much more effort into fixing this particular issue. In the event that we are stuck with it, @jgottula suggested a couple things:

- In gemini.py `exec_cmd()` and `exec_cmds()`, if in multiprocessing mode, wrap in a `multiprocessing.Lock()`. We may just be getting lucky that context switches are not happening right in the middle of this backend processing. This probably wouldn't fix the issue here though.
- Get backtrace on all processes at time it's blocked after first CTRL-C