seeing-things/track #225: context missing

vic4hub, 2021-03-29:
Hey guys a pretty interesting project, I've been thinking along the same lines though no idea where to start. I've got Celestron Evolution mount and ZWO 178 color version, trying to get this going on nvidia jetson nano 4gig. Would love to connect if at all possible.

bgottula, 2021-03-30:
Hi @vic4hub, thanks for the note. I suspect you would be the first aside from myself to try to use this code. It's still a work in progress, and I imagine there are some parts of the code that work fine on my hardware but which won't work as well on other hardware, but I will try to help as best I can. What are you hoping to accomplish?

vic4hub, 2021-03-30:
Hey @bgottula, ever since I looked through an eyepiece of a telescope over 6 years ago, I could not stop thinking about how cool it would be to survey the night sky for moving objects. In theory all one needs is an alt/azm electronic scope to be used as a fancy pan/tilt and some hardware for computer vision. I've played around with this idea for a few years now and ruled out some hardware, raspberry pi and different cameras for instance as not powerful/sensitive enough to do real time surveying. To sum this up I'm a bit stuck at getting the next iteration of hardware, namely jetson nano & zwo's implementation of Sony imx178 color, to play nice. Honestly I'm a bit disappointed with zwo software support given the price tag (the least they could do is make it work V4L).

bgottula, 2021-03-31:
Yeah I hear ya. Getting our ZWO 178 to perform well on a Linux machine ended up being quite a lot of work. Cameras that output RAW 

My main goal with this whole project has been to automate tracking of satellites. My vision for this is to be able to set up my hardware anywhere, prepare a list of satellites or other objects to track, hit "go" and have the software do everything except post process the raw imagery (alignment, tracking the objects, managing storage of the raw data). Not quite to full automation but the major pieces are there.

When you say you want to survey for moving objects, do you mean tracking objects that you know about ahead of time or objects that you find just by searching the sky?

vic4hub, 2021-03-31:
> When you say you want to survey for moving objects, do you mean tracking objects that you know about ahead of time or objects that you find just by searching the sky?
Why not both, theoretically there is tons of junk up there, as well as bunch of atmospheric phenomena that science cannot explain in its entirety (corona effects of sorts etc.) - why not use computer vision and some trained models to selectively pick out stuff automatically? Also post processing may not be required if you use a GPU.

closed 2021-09-24 by bgottula