seeing-things/track #274: `gps` package using deprecated `JSONDecoder` constructor arg `encoding`

bgottula, 2023-03-06:
The argument `encoding` has been deprecated since Python 3.1, [apparently](https://stackoverflow.com/a/67724070). Need to find a work-around for this. Is the `gps` package being actively maintained? I just refreshed my virtual environment so I'm using the latest version on PyPI, v3.19.

```
Traceback (most recent call last):
  File "/home/rgottula/dev/track/./track/stars.py", line 168, in <module>
    main()
  File "/home/rgottula/dev/track/./track/stars.py", line 135, in main
    location = gps_client.make_location_from_args(args)
  File "/home/rgottula/dev/track/track/gps_client.py", line 410, in make_location_from_args
    location = g.get_location(
  File "/home/rgottula/dev/track/track/gps_client.py", line 215, in get_location
    if self.client.read() == -1:
  File "/home/rgottula/dev/venv/lib/python3.10/site-packages/gps/gps.py", line 285, in read
    self.unpack(self.response)
  File "/home/rgottula/dev/venv/lib/python3.10/site-packages/gps/client.py", line 199, in unpack
    self.data = dictwrapper(json.loads(buf.strip(), encoding="ascii"))
  File "/usr/lib/python3.10/json/__init__.py", line 359, in loads
    return cls(**kw).decode(s)
TypeError: JSONDecoder.__init__() got an unexpected keyword argument 'encoding'
```

bgottula, 2023-03-06:
Okay so the `gps` package on PyPI is just an outdated version of the official Python client in the gpsd project. One of the maintainers listed on PyPI is also an active contributor to gpsd, so maybe that means it's "official" and not just some random person uploading it. In any case, I can see that the problem was fixed in the gpsd project in July 2020, but the most recent version of `gps` on PyPI is from 2019.

I opened https://gitlab.com/gpsd/gpsd/-/issues/232 in the hopes that the maintainers will push an update to PyPI. We'll see what they say.

bgottula, 2023-03-06:
Well unfortunately the gpsd developer whose name was listed as a maintainer on the PyPI package denies having any control or involvement except to complain to PyPA about it. Allegedly the PyPI package was in fact uploaded by someone else and is not actually associated with gpsd.

Alternatives I can think of:
- Try to email the supposed author of the PyPI package and see if that person is more helpful.
- Use the `python3-gps` package in Ubuntu, which probably has a more recent version that what's on PyPI and presumably is compatible with Python 3.10. But using that would mean that this package can no longer install that dependency automatically, which is unfortunate.
- Find some other gpsd client package on PyPI. I haven't come across one yet that looks particularly promising.
- Copy gpsd's Python client directly into this repo. Definitely not my favorite option but it would get the job done.
- Roll my own gpsd client. I only need a small subset of features, so this may not be terribly difficult. I could even start with the official client and refactor it to remove all the stuff I don't care about.

bgottula, 2023-03-06:
Some alternative packages on PyPI that have been updated within the past year or so:

- gpsdclient: Started in October 2021, last release in January. Very simple but looks competently designed, I like it.
   - https://pypi.org/project/gpsdclient/
   - https://github.com/tfeldmann/gpsdclient
- asyncio-gpsd-client: Started Nov 2021, last updated in September 2022. I don't think I need asyncio support, but I don't think that hurts anything.
   - https://pypi.org/project/asyncio-gpsd-client/
   - https://github.com/zachary822/asyncio-gpsd-client

jgottula, 2023-03-06:
I will see if I can do some digging on this...

jgottula, 2023-03-06:
> - gpsdclient: Started in October 2021, last release in January. Very simple but looks competently designed, I like it.
> 
>    - https://pypi.org/project/gpsdclient/
> 
>    - https://github.com/tfeldmann/gpsdclient

A quick peek suggests that this is probably the easiest good solution here. Looks like it was almost tailor-made for your "I just want a simple easy gpsd client" use case; and it's actually updated and so forth.

bgottula, 2023-03-07:
Okay thanks. That was my conclusion as well, so it's first on my list to try. Not sure when exactly I'll get to it, since replacing `gps` wasn't what I was actually planning on working on next. As a stop-gap measure I've simply edited the `gps` code installed in my venv to remove the deprecated constructor arg, which seems to have worked, so at least it's not truly blocking me in the near term.